nopnop2002/Arduino-STM32-CAN #17: Stm32f103 can bus not working

jwdsoft, 2020-04-14:
Hello
Can you help me figure out why this project is not working for me
It compiles successfully and get uploaded successfully by stlink swd connector
But doesn't send any messages 
The board doesn't work using usb doesn't get recognized so i can't see uart or debug info

nopnop2002, 2020-04-14:
Probably the cause is the value of the termination resistor.

When using a 3.3V CAN transceiver, the termination resistor value must be calculated exactly.

The easiest workaround is to use two ATmega and 5V transceivers with termination resistors, and the STM32 uses 3.3V transceivers without termination resistors.

zerog2k, 2020-08-06:
I do not think you can use CAN and USB on STM32F103 at same time. I think you will have to disable USB, and also I recall someone mentioning that the maple (old stm32 arduino framework enabled USB by default). I'm having issues with newer st-stm32 (stm32 duino, official newer arduino framework, and I suspect it's due to USB, but still troubleshooting this.)

nopnop2002, 2020-08-06:
My example does not use USB at all.

adamslodki, 2020-11-15:
My car have 3 CAN-BUS: first  33.3k, second 125K and  last 500k. It's possible to add 33.3k baudtate to your library? 
I analyzed the code. Another enum should be added:

enum BITRATE {CAN_33.3KBPS, CAN_50KBPS, CAN_100KBPS, CAN_125KBPS, CAN_250KBPS, CAN_500KBPS, CAN_1000KBPS};

and

CAN_bit_timing_config_t can_configs [7] = {{x, y, z}, {2, 13, 45}, {2, 15, 20}, {2, 13, 18}, {2, 13, 9}, {2, 15, 4}, {2, 15, 2}};

But what values to put in place of TS2 = x, TS1 = y, BRP = z?
[image]

closed 2020-11-17 by jwdsoft

maxkr2, 2022-01-18:
Hi,
I have the same issue on a STM32F103C8/ BluePill: USB is not working if CAN is active and not detected by Windows. CAN-transceiver is connected to PB8+9 as you told me in the past (issue #19) and I guess Serial.print() was possible via USB. 
Not sure whats wrong now. Hopefully you can give me a hint again.

nopnop2002, 2022-01-21:
When using Bluepill, STM32 core support for Arduino does not support onboard USB.

maxkr2, 2022-01-21:
Hi nopnop2002,
thank you for your detailed explaination. I was sure that I used integrated USB for serial output before, but using an serial-USB-converter at PA9+10 solved my problem: 
I got an output before also by integrated USB without CAN, I am also using CDC but on PlatfomIO and platform=Arduino. 
But with enabled CAN-transceiver the integrated USB was not detected in Windows any more (unknown device).

nopnop2002, 2022-01-21:
>with enabled CAN-transceiver the integrated USB was not detected

You are right.   
Select [Tools]-> [USB Support:CDC] to enable the virtual COM port.   
But CAN-transceiver and USB Support cannot be used at the same time.
Don't use USB Support.   
You have to use USB-Serial Converter.