nopnop2002/Arduino-STM32-CAN #23: How to add this files to arduino ide?

Marcelocm1995, 2020-07-15:
Hi! How to add this files to arduino ide?

nopnop2002, 2020-07-15:
Software requirements:

STM32 core support for Arduino.
https://github.com/stm32duino/Arduino_Core_STM32

Marcelocm1995, 2020-07-16:
I already have this "stm32duino" installed, where I put these folders of your library?

Sorry for my english. :)

nopnop2002, 2020-07-16:
This is sample code, not a library.

Compile immediately.

File -> Open -> stm32f103.ino

Marcelocm1995, 2020-07-16:
I understand. Now I have tried to compiled but output shows me "'CAN1' was not declared in this scope"

nopnop2002, 2020-07-16:
Are you useing this core library? (Not Maple core library)

STM32 core support for Arduino.
https://github.com/stm32duino/Arduino_Core_STM32

Marcelocm1995, 2020-07-21:
I was using another core, now I installed your core and compiles. Thanks!!!

nopnop2002, 2020-07-21:
It is good news for me.

closed 2020-07-21 by nopnop2002