nopnop2002/Arduino-STM32-CAN #34: Can Filters

Wh1terat, 2021-01-16:
Hi,
Firstly - thank you for the project, very nicely documented.

Just wondering if you have some examples for CANSetFilter ?

Going a little crazy here trying to get it to filter anything.

nopnop2002, 2021-01-17:
Hello.

0x681 << 21 = 0xd0200000

0x68103 << 13 = 0xd0206000

Check the CAN-ID

Wh1terat, 2021-01-19:
Thanks for reply.

Yes I understand it's 29bit for extended or 11bit for standard. 
But unlike mcp2515, stm32 filter seems to read from MSB, not MSB+3.

```text
11010000 00100000 00000000 00000000
```
Because this works to filter 0x681.

Infact, you can even see this in your own code from CANReceive function:

```c
      CAN_rx_msg->id = (CAN_STD_ID_MASK & (id >> 21U));
```

nopnop2002, 2021-01-19:
>is it possible to filter the first 2 data bytes of a standard frame using the 32bit filters?

The Filter bank register has 28 FiR1 and 28 FiR2 respectively.

F0R1-F27R1
F0R2-F27R2

Standard frame filter:

Set the first filter to F0R1
Set a second filter on F0R2

Wh1terat, 2021-01-19:
That is simply using list mode rather than mask mode to match multiple IDs ?

To explain my problem more clearly, Can ID 0x871 sends constant keepalive messages every 500ms, first byte of data ID is 0xF0.

I am only interested in messages where first byte of data is 0x04 - because of the frequency of the keepalive messages it makes ISR unsuitable.

With MCP2515's filter I am able to filter the ID and first 2 bytes of a standard frame, trying to replicate this functionality using STM32.

nopnop2002, 2021-01-19:
Standard frame filter Identifier List mode:

```
F0R1=first filter data
F0R2=second filter data
CANSetFilter(0, 0, 1, 0, F0R1, F0R2) 
```

Standard frame filter Identifier Mask mode:

```
F0R1=first filter data
F0R2=second filter data
CANSetFilter(0, 0, 0, 0, F0R1, F0R2) 
```

Wh1terat, 2021-01-19:
Thank you for trying to help, but I think maybe I am being unclear about my issue or what I am trying to achieve - or maybe it is not even possible.

Elektrik1, 2021-03-16:
Same issue here, transitioning from MCP2515, it is completely unclear how to configure CANSetFilter to allow particular (or range) of ID`s, also it's unclear how to filter on 0,1,2..n bytes of data, for example filter only particular (or range) where data[0] or data[n] corresponds.

nopnop2002, 2021-03-16:
To initialize the registers associated with the CAN filter banks (mode, scale, FIFO
assignment, activation and filter values), software has to set the FINIT bit (CAN_FMR). Filter
initialization also can be done outside the initialization mode.

advlogic, 2022-05-13:
I don't understand this either.  I want to filter only 7E8 diagnostic response messages from OBDII. 

No go even when directly editing the code as highlighted in the previous response:

CANSetFilter(0, 1, 0, 0, 0x7E8, 0x7E8);

nopnop2002, 2022-05-13:
0x7E8 is 24 bit.
You have to use Single 32-bit scale configuration.

nopnop2002, 2022-05-13:
[image]

```
  uint32_t bank1, bank2;
  bank1 = 0x7E8 << 21;
  bank2 = 0xFFE00000;
  CANSetFilter(0, 1, 0, 0, bank1, bank2);
```

v-ivanyshyn, 2022-12-21:
If anyone still struggles with setting up filters in list mode for 16bit IDs, here is how it should look:
uint32_t bank1 = (filterId1<<21) | (filterId2<<5);
uint32_t bank2 = (filterId3<<21) | (filterId4<<5);

rmoscoloni, 2023-03-10:
Please, could i ask how can i filter a range of std ids? ie 0 to 7?, list mode from v-ivanyshyn works well, but im getting confused by # if bitshift places. for mask mode.

I cant get why bank 2 is 0xFFE00006 In the example above:
 ```
 uint32_t bank1, bank2;
 bank1 = 0x7E8 << 21;
 bank2 = 0xFFE00006; // Must be IDE=0 RTR=0
 CANSetFilter(0, 1, 0, 0, bank1, bank2);
```

nopnop2002, 2023-03-10:
>I cant get why bank 2 is 0xFFE00006 In the example above:

Mask bit1 indicates that you care.

rmoscoloni, 2023-03-13:
Thanks for the answer nopnop2002:
first I have to thank you for the code you have made and shared, thank you very much!
Have the link bootmarked for a long time and I wanted to follow the example of filtering from 0 to 7 but in std id.
I was trying to understand how to include in your code to filter the mentioned range.

Answering my own question and for reference to future readers, 
0xFFE00006 --> 0x7ff << 21 + 0x06
11111111111000000000000000000000+
00000000000000000000000000000110=
11111111111000000000000000000110

why include 0x06 in the filter? 0x7ff<<21 is not enough?

nopnop2002, 2023-03-13:
>why include 0x06 in the filter? 0x7ff<<21 is not enough?

This is because IDE and RTR are used to distinguish between packet types.

```
Bit 2 IDE: Identifier extension
This bit defines the identifier type of message in the mailbox.
0: Standard identifier.
1: Extended identifier.
Bit 1 RTR: Remote transmission request
0: Data frame
1: Remote frame
```

martin323232, 2023-10-31:
Having issues with filter settings,

receiving id 0x101 and 0x206, how to setup a filter to just receive these messages.

Tried :        
```
uint32_t bank1 = (0x101<<21) | (0x206<<5);
uint32_t bank2 = (0x0<<21) | (0x0<<5);

CANSetFilter(0, 0, 1, 0, bank1, bank2);
```
```
uint32_t bank1 = (0x101<<21) | (0x101<<5);
uint32_t bank2 = (0x206<<21) | (0x206<<5);

CANSetFilter(0, 0, 1, 0, bank1, bank2);
```

nopnop2002, 2023-10-31:
```
CAN1->FMR  |=   0x1UL;                // Set to filter initialization mode

fileter setting

CAN1->FMR   &= ~(0x1UL);              // Deactivate initialization mode
```