nopnop2002/Arduino-STM32-CAN #35: sent fail issue

amoljamble1, 2021-02-04:
[image]
getting this issue with stm32f103

nopnop2002, 2021-02-04:
read Troubleshooting in README.

saadat925, 2021-04-13:
Dear sir

I have encountered this error message while running the stm32f103 canbus program, please help

nopnop2002, 2021-04-13:
There are three possible causes

1. The receiving app is not running.

2. The speed does not match the receiving side.

3. 3. The resistance value of the terminating resistor is incorrect.

saadat925, 2021-04-14:
Hi 
I get this error message while compiling the program, please help me.
[image]
I also use this core :

saadat925, 2021-04-14:
I also use this core
[image]

nopnop2002, 2021-04-14:
>I also use this core

I don't know what core you use.

I don't know what error you get.

saadat925, 2021-04-14:
[image]
I used this core

nopnop2002, 2021-04-14:
Drag & drop png file here.

saadat925, 2021-04-14:
[image]

saadat925, 2021-04-14:
[image]

nopnop2002, 2021-04-14:
Before post, preview your post.

saadat925, 2021-04-14:
[image]

saadat925, 2021-04-14:
[image]

nopnop2002, 2021-04-14:
I can't see.
Can you see your png file?

saadat925, 2021-04-14:
[Uploading error.pdf…]()

nopnop2002, 2021-04-14:
https://stackoverflow.com/questions/10963205/how-to-attach-file-to-a-github-issue

nopnop2002, 2021-04-14:
＞I also use this core

What core do you use?

saadat925, 2021-04-14:
Do you have any email ?

nopnop2002, 2021-04-14:
The github problem can only be solved here.

saadat925, 2021-04-14:
[error.pdf](https://github.com/nopnop2002/Arduino-STM32-CAN/files/6309692/error.pdf)

saadat925, 2021-04-14:
I was finally able to send the file, please check and help.

nopnop2002, 2021-04-14:
Yes I can see your file.

The core you are using does not support CAN.

saadat925, 2021-04-14:
Excuse me which readme?

saadat925, 2021-04-14:
I can not use the blue pill board?

nopnop2002, 2021-04-14:
https://github.com/nopnop2002/Arduino-STM32-CAN/blob/master/README.md

>I can not use the blue pill board?

No. 
You can use it.
But you have to change core library.

saadat925, 2021-04-14:
Where can I find this core library?

nopnop2002, 2021-04-14:
Software requirements
STM32 core support for Arduino.
https://github.com/stm32duino/Arduino_Core_STM32

saadat925, 2021-04-14:
Do you mean the "arduino_core_stm32_master" library?

nopnop2002, 2021-04-14:
There is some core library.

You are using:
https://github.com/rogerclarkmelbourne/Arduino_STM32

This require:
https://github.com/stm32duino/Arduino_Core_STM32

saadat925, 2021-04-14:
thank you sir

saadat925, 2021-04-14:
Excuse me 
Which file and how should I use?
[Uploading core.pdf…]()

saadat925, 2021-04-14:
[core.pdf](https://github.com/nopnop2002/Arduino-STM32-CAN/files/6310048/core.pdf)

saadat925, 2021-04-14:
Which file and how should I use?

saadat925, 2021-04-14:
Hi
How to add these cores in offline mode?
[core.pdf](https://github.com/nopnop2002/Arduino-STM32-CAN/files/6310401/core.pdf)

nopnop2002, 2021-04-14:
Follow this document
https://github.com/stm32duino/Arduino_Core_STM32

saadat925, 2021-04-14:
Hi
I want to to program to board by st-link programm
[image]
er, please help with the settings.

nopnop2002, 2021-04-14:
Blue-pill STM32 Programming Using ST-LINK V2 Dongle

https://www.youtube.com/watch?v=KgR3uM21y7o

saadat925, 2021-04-15:
Thank you sir.