nopnop2002/Arduino-STM32-CAN #38: convert to user-friendly library

s54b32julian, 2021-04-03:
hello, first thanks for your work!
Maybe for usability and to simplify the code for the enduser, convert to a library? So that the main code dont get overload with background tasks. Only include the lib, init the CAN and call constructors like ReadCan or SendCan in the loop.

nopnop2002, 2021-04-03:
As you can see from the source, the BxCan registers are different for each model.

Even if it is converted to a library, it will be a library for each model.

s54b32julian, 2021-04-04:
Maybe you could merge all in one library and the user make config with #define at the begin of the sketch.

like 
`
#define MCU F40x //MCU variant 
#define Channels 1 //1 or 2 available channels
#define RX1 PA10 //RX pin Channel 1
#define TX1 PA9  //TX pin Channel 2
`

nopnop2002, 2021-04-04:
F103 and others are too different.
Almost impossible to merge,

seb43654, 2022-05-08:
hi, are you able to create an example arduino program for one IC, then just change the included .h file? I am struggling to understand how to use your code

nopnop2002, 2022-05-08:
What is one IC?

nopnop2002, 2023-09-10:
I've ported the code to a number of STM32s.

As a result, it turned out that it is completely impossible to support many boards with one library.

closed 2023-09-10 by nopnop2002