nopnop2002/Arduino-STM32-CAN #39: Sample program

saadat925, 2021-04-18:


saadat925, 2021-04-18:
Dear friend
I want to send 0x01 to identifier 0 only once and then receive 8 bytes data from id = 2bd permanently.  please guide me .  I also used a blue pill.  Thank you

nopnop2002, 2021-04-18:
This is simple example for blue  pill.

https://github.com/nopnop2002/Arduino-STM32-CAN/tree/master/stm32f103

saadat925, 2021-04-18:
I run this program and get
[image]
this result.  But I need to send a byte

nopnop2002, 2021-04-18:
You need to change this code.

https://github.com/nopnop2002/Arduino-STM32-CAN/blob/master/stm32f103/stm32f103.ino#L360

saadat925, 2021-04-24:
Dear sir
Now the canbus is sent on my device in extended mode. If I want to be sent only in standard mode, what do you suggest?

nopnop2002, 2021-04-24:
change this.

https://github.com/nopnop2002/Arduino-STM32-CAN/blob/master/stm32f103/stm32f103.ino#L376

saadat925, 2021-04-24:
That is, to replace the extended_format with the standard_ format ?

nopnop2002, 2021-04-24:
Yes.

saadat925, 2021-04-24:
Where can I find UNO program?

nopnop2002, 2021-04-25:
I can't understand your question.

closed 2021-10-03 by nopnop2002