nopnop2002/Arduino-STM32-CAN #40: Use on STM32F405RGT6 custom board

advlogic, 2021-07-06:
I am getting send fails on a custom board using the STM32F405 code.  I had everything working on a Bluepill STM32F103.  It isn't the terminating resistors.  If I am using an external HSC XTAL, do I have to change anything in the code?

I can use the same test rig (Microchip CAN Analyzer, harness, resistors, etc) on my bluepill based prototype and it works fine.

I can't think of anything else.  I am using CAN1 only on PB8 and PB9.   Transceiver is SN65HVD230.  The code initializes it fine in the CANinit function.

// CAN1
    RCC->AHB1ENR |= 0x2;                 // Enable GPIOB clock 
    CANSetGpio(GPIOB, 8);                // Set PB8
    CANSetGpio(GPIOB, 9);                // Set PB9

All my other peripherals (TFT, I2C EEPROM) all work fine.  Here is my HAL clock config that I used STM32CubeIDE to find.

extern "C" void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

  /** Configure the main internal regulator output voltage
  */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);
  
  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI;
  RCC_OscInitStruct.PLL.PLLM = 8;
  RCC_OscInitStruct.PLL.PLLN = 168;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 4;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }
  /** Initializes the CPU, AHB and APB buses clocks
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK)
  {
    Error_Handler();
  }
}

advlogic, 2021-07-06:
My transceiver circuit:

[image]

Pins on the micro:

[image]

advlogic, 2021-07-06:
From my RTT serial:

00> GPIO_AFR(b)=0
00> GPIO_AFR(a)=9000
00> GPIO_MODER(b)=A800A915
00> GPIO_MODER(a)=A880A915
00> GPIO_OSPEEDR(b)=C00FF3F
00> GPIO_OSPEEDR(a)=CC0FF3F
00> GPIO_OTYPER(b)=0
00> GPIO_OTYPER(a)=0
00> GPIO_PUPDR(b)=64005800
00> GPIO_PUPDR(a)=64005800
00> GPIO_AFR(b)=9000
00> GPIO_AFR(a)=99000
00> GPIO_MODER(b)=A880A915
00> GPIO_MODER(a)=AA80A915
00> GPIO_OSPEEDR(b)=CC0FF3F
00> GPIO_OSPEEDR(a)=FC0FF3F
00> GPIO_OTYPER(b)=0
00> GPIO_OTYPER(a)=0
00> GPIO_PUPDR(b)=64005800
00> GPIO_PUPDR(a)=64005800
00> GPIO_AFR(b)=0
00> GPIO_AFR(a)=9
00> GPIO_MODER(b)=A280
00> GPIO_MODER(a)=2A280
00> GPIO_OSPEEDR(b)=F0C0
00> GPIO_OSPEEDR(a)=3F0C0
00> GPIO_OTYPER(b)=C0
00> GPIO_OTYPER(a)=C0
00> GPIO_PUPDR(b)=100
00> GPIO_PUPDR(a)=100
00> GPIO_AFR(b)=9
00> GPIO_AFR(a)=99
00> GPIO_MODER(b)=2A280
00> GPIO_MODER(a)=AA280
00> GPIO_OSPEEDR(b)=3F0C0
00> GPIO_OSPEEDR(a)=FF0C0
00> GPIO_OTYPER(b)=C0
00> GPIO_OTYPER(a)=C0
00> GPIO_PUPDR(b)=100
00> GPIO_PUPDR(a)=100
00> CAN1->MCR=10003
00> CAN2->MCR=10003
00> CAN1->BTR=1A0005
00> CAN2->BTR=1A0005
00> CAN1->FMR=2A1C0001
00> CAN Channel 2: Not Available
00> CAN Channel 1: Initialized
00> CAN transciever initialized
00> Send fail CAN channel 1

advlogic, 2021-07-06:
I don't understand how CANSetGpio sets appropriate pins for RX and TX either.  Thanks for the help!

nopnop2002, 2021-07-06:
If you change CPU Frequence, you need to change the code here

https://github.com/nopnop2002/Arduino-STM32-CAN/blob/master/stm32f407/stm32f407.ino#L29

Maybe you can use this for CAN Bit Time Calculation.

http://www.bittiming.can-wiki.info/

advlogic, 2021-07-07:
Thank you for your reply.  I have some more studying to do on bit timings.  What values do I enter into this calculator?

I can see some activity on the CAN Bus Analyzer but am receiving many errors.  I believe it is indeed a timing issue.

advlogic, 2021-07-07:
[image]

advlogic, 2021-07-07:
Seems to match up with the defaults for 500kbps in the source.

[image]

advlogic, 2021-07-07:
The default timing values were correct even with my HSE XTAL.  The issues were from the CAN2 being assigned to floating pins.  I removed all references to CAN2 in the code as I am not ever going to use dual CAN bus.

closed 2021-07-07 by advlogic