nopnop2002/Arduino-STM32-CAN #41: Problem when compiling stm32f103.ino

mamech, 2021-07-28:
Hello

I appreciate the effort done in this project. I have a problem with compilation of  stm32f103.ino 
I got this error:
"C:\Users\XXXXX\Documents\Arduino\libraries\stm32f103\stm32f103.ino: In function 'void CANReceive(CAN_msg_t*)':
stm32f103:242:8: error: 'id' was not declared in this scope
  242 |   if ((id & STM32_CAN_RIR_IDE) == 0) { // Standard frame format
      |        ^~
stm32f103:251:8: error: 'id' was not declared in this scope
  251 |   if ((id & STM32_CAN_RIR_RTR) == 0) { // Data frame
      |        ^~
exit status 1
'id' was not declared in this scope"
I have Arduino IDE 1.8.12
and latest version of STM32duino

mamech, 2021-07-28:
[](url)

mamech, 2021-07-28:
[image]
This is the configuration I selected.

nopnop2002, 2021-07-28:
Your code doesn't have this line.

https://github.com/nopnop2002/Arduino-STM32-CAN/blob/master/stm32f103/stm32f103.ino#L243

mamech, 2021-07-29:
Thank you very much. now compiled successfully.
But I faced a new problem. The program gets stuck in Setup() function and does not continue to loop() function for unknown reason.
`
void setup() {
  Serial.begin(115200);
  Serial.println("Hi 1");
  ///bool ret = CANInit(CAN_500KBPS, 0);  // CAN_RX mapped to PA11, CAN_TX mapped to PA12
  bool ret = CANInit(CAN_100KBPS, 3);  
   //bool ret = CANInit(CAN_100KBPS, 2);  // CAN_RX mapped to PB8, CAN_TX mapped to PB9
  //bool ret = CANInit(CAN_1000KBPS, 0);  // CAN_RX mapped to PA11, CAN_TX mapped to PA12
  //bool ret = CANInit(CAN_1000KBPS, 2);  // CAN_RX mapped to PB8, CAN_TX mapped to PB9
  //bool ret = CANInit(CAN_1000KBPS, 3);  // CAN_RX mapped to PD0, CAN_TX mapped to PD1
   Serial.println("Hi 2");
  if (!ret) while(true);
  Serial.println("Hi 3");
}
`

Only "Hi 1" is printed. so this means to me that this line  :
bool ret = CANInit(CAN_100KBPS, 3); 
is blocking the code for some reason, so code stucks inside CANInit() and does not come out of it. I tried to change the second argument of CANInit() to 0,1,2 and all are the same, the program stucks.
I am using Bluepill.

nopnop2002, 2021-07-29:
Possible reasons

bluepill does not have a CAN transceiver.

mamech, 2021-07-29:
I think you mean the blue pill was not connected to CAN transceiver during this test?
The answer is yes, but blue pill itself has CAN controller support, so why do I need a transceiver? And how a hardware transceiver causes the code to get stuck?

nopnop2002, 2021-07-29:
blue pill itself has CAN controller support.
you need CAN transceiver.

Read the README

mamech, 2021-08-02:
Thanks for clarification. I tried to make like echo test using one blue pill and transceiver, but did not work either. I am wondering is it a must to have 2 blue pills with 2 transceivers? can not I use only onle blue pill with CANHigh connected to CANLow so I receive the same message I sent?

nopnop2002, 2021-08-02:
it a must to have 2 blue pills with 2 transceivers.

CAN is different from UART.

It is expressed by the voltage difference between CAN HIGH and CAN LOW.

https://www.ti.com/lit/an/slla337/slla337.pdf

mamech, 2021-08-02:
Thanks, now it is clear. It is first time I use CAN so I am influenced much with UART.
One more question please, CAN network should have master and slave, so these 2 bluepills should be configuered to be one master and the other is slave, so this leads to 2 questions:
1- The code we are talking about is made to make its bluepill a master or slave?
2- If it is a master how to reverse its function to be a slave (or vice versa) ?
Thanks

nopnop2002, 2021-08-02:
There is no master / slave in the CAN network.

There is just a sender / receiver.

Anyone can be a sender.
Anyone can be a receiver.

closed 2021-10-03 by nopnop2002