nopnop2002/Arduino-STM32-CAN #43: Adapt to stm32f767 custom board

0v3rcl0ck3d, 2022-04-24:
Hello nopnop2002,
Thanks for your create work.
I want do adapt your code to a board using the stm32f767 but I don't know how to get this done. Could you provide me some information which code is specific for the processor? Where did you get the information to support the other processors?

nopnop2002, 2022-04-24:
You need to read the Reference manual carefully.

https://www.st.com/resource/en/reference_manual/dm00224583-stm32f76xxx-and-stm32f77xxx-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf

nopnop2002, 2022-05-04:
>From what I saw in the manual, it seems that the CAN part of the 767 and the 466 are nearly the same. 

It looks the same at first glance, but it may be slightly different.

It is desirable for the STM to support generic libraries.

0v3rcl0ck3d, 2022-05-04:
If you just need a 767 and could help me with it, this should be doable. However, we are short on time and need a working solution within days. If there is a way to achieve this, let me know.

nopnop2002, 2022-05-04:
>we are short on time and need a working solution within days. 

It's a lot of work.
It is impossible in a short time.

nopnop2002, 2023-09-03:
Added F7 series

closed 2023-09-03 by nopnop2002