nopnop2002/Arduino-STM32-CAN #44: hi,my board is stm32f103zet6,have some issues

ChinaBurn, 2022-05-18:
when i use CANInit(CAN_500KBPS, 3);
always return false;
i guess can init failed.
the same issue when i change parameter,like CANInit(CAN_1000KBPS, 3); or CANInit(CAN_500KBPS, 2);

can you help? thanks

nopnop2002, 2022-05-18:
If the connection is not correct, it will not return to normal.

ChinaBurn, 2022-05-19:
> If the wiring with the CAN Transceiver is not correct, it will not return to normal.

I'm pretty sure the Can transceiver is wired correctly because I'm using the HAL's code and the Can is functioning properly. Ability to send and receive Can frames。but i love stm32duino~,and i wang find out where is the problem

ChinaBurn, 2022-05-19:
> Does this work correctly
> 
> CANInit (CAN_500KBPS, 1) CANInit (CAN_1000KBPS, 1)

Nope,can't work.Is the CAN transceiver necessary during the CAN initialization process?

nopnop2002, 2022-05-19:
>Is the CAN transceiver necessary during the CAN initialization process?

Yes.

ChinaBurn, 2022-05-19:
I tested STM32F103VET. F103VET has PD0 and PD1. The data sheet has the same specifications for F103xC, F103xD, and F103xE.

Your STM32F103VET is OK?
My STM32duino ver is 2.2.0 by STMicroelectronics

nopnop2002, 2022-05-19:
>Your STM32F103VET is OK?

Yes

Try changing the Board part number to VET.

It only changes the maximum GPIO available.

ChinaBurn, 2022-05-19:
> > 您的STM32F103VET可以吗？
> 
> 是的
> 
> 尝试1）不要使用USB支持。基本扩展 CAN （BxCan） 和 USB 支持不能同时使用。这在我的自述文件中进行了描述。
> 
> Try2） 将整体选件板的部件号更改为 VET。
> 
> 它仅更改可用的最大 GPIO。

thanks very much.

ChinaBurn, 2022-05-25:
There was some progress until today that I found to be related to **the board part number.**
I have always selected "Generic F103ZETX" before, and have never successfully communicated.

I tried "VCCGND F103ZET6", CANInit(CAN_1000KBPS, 0) and CANInit(CAN_1000KBPS, 1) can communicate successfully. But CANInit (CAN_1000KBPS, 3) is initialized successfully and communication fails.

I feel that this has something to do with the framework of STM32duino.

nopnop2002, 2022-05-25:
The cause is the difference between the two.

https://github.com/stm32duino/Arduino_Core_STM32/blob/main/variants/STM32F1xx/F103Z(C-D-E)(H-T)/variant_generic.h

https://github.com/stm32duino/Arduino_Core_STM32/blob/main/variants/STM32F1xx/F103Z(C-D-E)(H-T)/variant_VCCGND_F103ZET6.h

nopnop2002, 2022-05-25:
This may be relevant.

This function is only ARDUINO_VCCGND_F103ZET6.

https://github.com/stm32duino/Arduino_Core_STM32/blob/main/variants/STM32F1xx/F103Z(C-D-E)(H-T)/variant_VCCGND_F103ZET6_XXX.cpp#L168

ChinaBurn, 2022-05-25:
It is beyond my ability to solve this problem at present, so I can only put it aside temporarily.

nopnop2002, 2022-05-25:
Probably the clock frequency is not set to 72MHz.

You can find your board clock.

Serial.println(F_CPU);

ChinaBurn, 2022-05-25:
> Probably the clock frequency is not set to 72MHz.
> 
> You can find your board clock.
> 
> Serial.println(F_CPU);
> 
> This is the result of Blue Pill. 72000000

I have checked the clock frequency and it is confirmed to be 72MHz.
but PD0,PD1 still can't use on "VCCGND F103ZET6"

nopnop2002, 2022-05-25:
Ok.

>It is beyond my ability to solve this problem at present

I'm too.

nopnop2002, 2022-07-13:
Hello.

There is solution.

https://github.com/nopnop2002/Arduino-STM32-CAN/issues/45

ChinaBurn, 2022-07-13:
> 你好。
> 
> 有解决方案。
> 
> #45
Thanks,I'will test it tomorrow

nopnop2002, 2022-08-21:
@ChinaBurn 

Added support for Generic F103 board.

The CPU clock frequency of Board Part Number BluePill and BlackPill is 36Mhz, but Generic Board is 32Mhz.

The latest version also works at 32Mhz.

ChinaBurn, 2022-08-22:
> @ChinaBurn
> 
> 添加了对通用 F103 板的支持。
> 
> Arduino内核对基于STM32的电路板的支持使用不同的时钟频率，具体取决于电路板部件号。 BlackPill & BluePill使用72Mhz，通用板使用64Mhz。
> 
> 最新版本适用于72Mhz和64Mhz。

Thank you very much ten thousand times

ChinaBurn, 2022-08-22:
can you give me your email address?thanks

geraldjust, 2022-08-28:
Just wanted to let you know, reason that the F103 chips cant use USB and CAN share the the memory buffers. But F0,F3,F4,F7 can. 

https://jeelabs.org/2018/canbus-part-3/

nopnop2002, 2022-08-28:
@geraldjust 

Thank you.