nopnop2002/Arduino-STM32-CAN #46: Send Fail

ciamia, 2022-08-19:
Hi, ı want to receive messages from arduino mega with canbus on stm32f407. When upload code it says CAN initilization is succesfull but doesnt receive anything. Arduino prints sending is succesfull. But I cant figure out why it doesnt work. And I counldnt find your Sender Codes anywhere, are they avaliable?

nopnop2002, 2022-08-19:
It is work with stm32F405/406.

https://github.com/nopnop2002/Arduino-STM32-CAN/tree/master/stm32f407

ciamia, 2022-08-19:
I used your code but I am getting send fail on my screen. I tried the things you said and they didnt seem to work for me. My two arduinos talk to each other just fine but when I add stm32 with your code my receiver doesnt get anything and yours say send fail.

nopnop2002, 2022-08-19:
If the transmission fails, these are the possible causes.

There is no receiving app on CanBus.
The speed does not match the receiver.
There is no terminating resistor on the CanBus.
There are three terminating resistors on the CanBus.
The resistance value of the terminating resistor is incorrect.
Stub length in CAN bus is too long.
Though the ISO11898 Standard specifies a maximum bus length of 40m and maximum stub length of 0.3m at 1Mbps. See [here](https://e2e.ti.com/support/interface-group/interface/f/interface-forum/378932/iso1050-can-bus-stub-length).

nopnop2002, 2022-08-21:
Your UART-USB converter connection may be wrong.

Make sure you can see Serial.print using this.

```
void setup() {
  Serial.begin(115200);
  while (!Serial) ;
  Serial.println("STM32 Hello start");
}

void loop() {
  Serial.print("Hello from STM32! ");
  Serial.print(F_CPU);
  Serial.print(" ");
  Serial.println(millis());
  delay(1000);
}
```

ciamia, 2022-08-22:
I am going to try the cables. Do you have the code for arduino receiver. Did you share it somewhere? Maybe I can try that.

ciamia, 2022-08-22:
And I tried the code you said it works perfectly. My UART-USB converter connection is fine.

ciamia, 2022-08-22:
My problem isnt solved. Only Serial connection is ok. But communication problem continious.

ciamia, 2022-08-22:
> 

?

ciamia, 2022-08-22:
I really need to try the arduino code you used.

nopnop2002, 2022-08-22:
https://github.com/nopnop2002/Arduino-STM32-CAN/blob/master/README.md

ciamia, 2022-08-22:
Sir, I am trying to send data from Mega and Stm32 then read it from Nano. Mega and Nano works perfectly together but when I connect Stm32 with your code. Nano stops receiving and Stm32 says send fail. I connected resistors and everyting. I am asking if you shared the code that you used in Arduino to try it. I have read the README.

nopnop2002, 2022-08-22:
>I am asking if you shared the code that you used in Arduino to try it. 

https://github.com/coryjfowler/MCP_CAN_lib

nopnop2002, 2022-08-22:
1000KBPS is susceptible to noise.

Try it at 250KBPS or 500KBPS.

nopnop2002, 2022-08-22:
Mega and Nano are working perfectly, what happens when you add STM32?

But without turning on the STM32.

ciamia, 2022-08-22:
I solved the problems they work now thank you for your help. I used the arduino codes you suggested and changed the clock settings. It works perfectly now.

nopnop2002, 2022-08-23:
I found that:

STM32F407 can use a Virtual COM port.
If you enable Virtual COM port, you can use the USB port as a serial port.
Basic Extended CAN(BxCan) and USB Support can be used at the same time.

ciamia, 2022-08-23:
Yes, I used USB as a serial port

closed 2022-08-23 by nopnop2002