nopnop2002/Arduino-STM32-CAN #48: Possible to read but can't send data.

Toernblom, 2022-09-22:
Hi, currently using STM32F103R8T6.

CAN interface is connect like: 
PB8 -> CAN_RX -> TJA1050
PB9 -> CAN_TX -> TJA1050

in void Setup I'm using:
bool ret = CANInit(CAN_500KBPS, 2); // CAN_RX mapped to PB8, CAN_TX mapped to PB9

What happens is that while debugging i notice that on [this ](https://github.com/nopnop2002/Arduino-STM32-CAN/blob/1fe73447a522e0dd3f3294d88ab8a86e79e69ae4/stm32f103/stm32f103.ino#L481)line, the while loop never ends. 

Communication works ok outside this unit. Am i missing something?

Toernblom, 2022-09-22:
[image]

Errors while running. I've used the same cables with another board using CAN so shouldn't be an issue. Terminations is ok aswell because other devices can talk to each other.

nopnop2002, 2022-09-22:
Regarding send fail, past issues are helpful.

Toernblom, 2022-09-23:
I currently have 3 nodes, 2 of them are talking and all has twisted shielded pairs which are 5-10cm in length and the shield is grounded. It's the only device on the network that can't talk. The resistance between CAN_L & CAN_H is ~60 Ohm so that isn't an issue either. I've also checked previous closed issues and tried those aswell. All baudrates is 100kbps to minimize noise. 

[image]

These are the prints from your code. I noticed that the peripheral_clock_rate was at 32mhz while STM32F103R8 that i'm using is at 72mhz so i tried changing that aswell to the correct clock with no success

nopnop2002, 2022-09-23:
peripheral_clock_rate is not cpu frreuency.

If the receive works correctly, it's not a CPU frequency issue.

Toernblom, 2022-09-23:
Hmm okey, then peripheral_clock_rate should be 36mhz then? Since STM32F103R8T6 is 72Mhz not 64. 

The receive did work yesterday, not anymore for some reason.. I saw your previous comment on another issue with 150 Ohm termination, I'll try that!

nopnop2002, 2022-09-23:
>Since STM32F103R8T6 is 72Mhz not 64.

Arduino core support for STM32 based boards uses different clock frequencies depending on the Board Part Number.
BlackPill & BluePill use 72Mhz.
Generic board use 64Mhz.

Toernblom, 2022-09-23:
Hmm, that result gave 64 Mhz. In my Platformio.ini file , the board is genericSTM32F103R8.
Should i switch out the board part number?

nopnop2002, 2022-09-23:
>Should i switch out the board part number?

If you use bluepill or blackpill  in Platformio.ini, it will be 72Mhz.

It has nothing to do with this issue.

Toernblom, 2022-09-23:
Haha i understand why. We use a custom made PCB with STM32F103R8T6 and It's the first time with the STM32 chip family. So I thought i'd should use the genericSTM32F103R8 in the .ini file , not blackpill or bluepill ..

nopnop2002, 2022-09-23:
I think you can build custom boards using BluePill and BlackPill variants.

However, the GPIO of the Serial object changes for the Generic board, BluePill, and BlackPill.

everyone gets confused.

Toernblom, 2022-09-23:
Can't use bluepill or blackpill in platformio.ini file to make serial work. If i use genericSTM32F103R8 with HardwareSerial its fine.
[image]
its because 
[image]

However, if i do use genericSTM32F103R8 F_CPU is wrong and if i use either bluepill or blackpill in .ini the f_CPU is correct..

I do understand while everyone gets confused..

Toernblom, 2022-09-23:
I have replaced the termination resistors with 150 Ohm so that i have two terminations at the ends with a bus resistance at 75 Ohm now. No success in sending any data.

nopnop2002, 2022-09-23:
Keep STAB length as short as possible.

Toernblom, 2022-09-23:
[image]

This is the setup, the connection between the raspberry pi on the right and the CAN analyze tool on the bottom has no shielded nor twisted pairs. I'm not sure the cables are the issues..

nopnop2002, 2022-09-23:
If you have a BluePill or BlackPill, you can identify the problem.

Toernblom, 2022-09-23:
Sadly I don't. I can purchase one however if you think that would help ?

nopnop2002, 2022-09-23:
There may be a problem with your custom board.

If it works fine with BluePill/BlackPill, then there may be a problem with your custom board.

Toernblom, 2022-09-23:
The connection:
[image]
[image]

[image]

Your code says that the initialization was ok. I don't think theres anything wrong with the wiring here..

Toernblom, 2022-09-23:
Alright, found out the reason why reading stop worked. It was cause i was monitoring on the UART. If i only powered it on and ran with debug i could read messages. Not send however. Its quite weird actually.

If dont use Hardware Serial at PA10 & PA9

Then the code gets stuck here
`CANInit(CAN_250KBPS, 2);`

Stepping over CANInit function with debugging results in the debug can't step over and gets stuck inside this function
`ComputeCANTimings(HAL_RCC_GetPCLK1Freq(), target_bitrate, &timings);`

Digging further it goes to `stm32f1xx_hal_rcc.c`
and gets stuck here

```
uint32_t HAL_RCC_GetPCLK1Freq(void)
{
  /* Get HCLK source and Compute PCLK1 frequency ---------------------------*/
  return (HAL_RCC_GetHCLKFreq() >> APBPrescTable[(RCC->CFGR & RCC_CFGR_PPRE1) >> RCC_CFGR_PPRE1_Pos]);
}
```

And if i do use Hardware Serial at PA10 & PA9
the code doesn't hang on that line and continues running and i can successfully read messages on the CAN bus. But CANSend hangs here: 

`  while (CAN1->sTxMailBox[0].TIR & 0x1UL && count++ < 1000000);`

nopnop2002, 2022-09-24:
This is the UART output.
Something is wrong.
[image]

Toernblom, 2022-09-24:
Alright, If i put a delay between Serial.begin and CANInit the code doesn't hang anymore. Still getting Send Fail no matter what i do, I've got another CAN Node with TJA1050 exact same cables and length on the network that is working well, custom made PCB aswell..

amitesh-singh, 2024-03-27:
@Toernblom have you found the solution?

nopnop2002, 2024-03-27:
@Toernblom 

It's not a clone chip, right?

https://hackaday.com/2020/10/22/stm32-clones-the-good-the-bad-and-the-ugly/

Toernblom, 2024-03-30:
Hi, it was a fault within the PCB. I think it was some issue with different logic levels between the IC's of the MCU and CAN controller & transciever.

amitesh-singh, 2024-03-30:
@Toernblom @nopnop2002 Thanks for the reply.

Is it due to powering tja1050 by 5v and stm32f103 is powered from 3.3v?

amitesh-singh, 2024-03-30:
[image]
[image]

cc @Toernblom @nopnop2002

amitesh-singh, 2024-03-30:
the PB8 and PB9 pins are 5v tolerant and tja1050 should work with 3.3v devices, that's what they claim in their datasheet as tja1050 RXD is a totem pole and voltage varies from 0.7Vcc to 0.45v

nopnop2002, 2024-03-30:
@amitesh-singh 

+1V Ground Shift with Split Termination is effective when CAN-BUS has a mix of 3.3V and 5V transceivers.