nopnop2002/Arduino-STM32-CAN #52: STM32F103 gets stuck inside while (!(CAN1->MSR & 0x1UL));

MaxYoshida, 2023-01-20:
Hello

This is amazing, and I have been trying to make it work on STM32F103C8T6 for a month now using this code and other codes as well, but with no success. So I believe it is time now to ask for help.

Issue is simple: I flash the code with Serial.print debugs and it gets stuck inside loop while (!(CAN1->MSR & 0x1UL));

Any ideas what could be wrong? Why is the MSR not set in CAN1->MCR |= 0x1UL; ?

Would really appreciate your support!

MaxYoshida, 2023-01-20:
Just to be very clear and avoid possible misunderstandings: I made this modification to the code of STM32F103:

Serial.println("IN1");
  CAN1->MCR |= 0x1UL;                   // Require CAN1 to Initialization mode 
  Serial.println("IN2");
  while (!(CAN1->MSR & 0x1UL));         // Wait for Initialization mode
  Serial.println("IN3");

When running it, on Serial I get both IN1 and IN2, but IN3 is never reached..

nopnop2002, 2023-01-20:
Are you using Virtual COM ports?

MaxYoshida, 2023-01-20:
I am not sure what you mean exactly. The only modification I did to the code you provided was adding Serial.println commands to see where it gets stuck. I use ST-LINK V2 to upload the code (SWD) and I use a USB TTL converter on pin PA9 and PA10 to get Serial information.

May I suggest that I pay you for a debug session? Name the price and your means of transaction. I am willing to pay for your time on return of getting it working. A lot of things here depend on it!

nopnop2002, 2023-01-20:
>I use a USB TTL converter on pin PA9 and PA10 to get Serial information.

OK.

I want to know your environment.
- Compile option
- core library version

MaxYoshida, 2023-01-20:
About core library: I always use latest. I have STM32 MCU based boards library by STMelectronics V2.4.0 where the code compiles correctly, and stm32duino V2022.9.26 but the code does not compile in it and it says "CAN1 undefined"

About transceiver, I have MCP2551 module and SN65HVD230 module. Of course when testing I do not mix them up. However, as you noticed, my problem in code is before transmission even begins to happen.

Compile option: I left everything as default from library. Only thing I made sure of is that UART support is enabled and USB support is set to None.

Tell me how to consider the paid debug session?

nopnop2002, 2023-01-21:
I use STMelectronics V2.4.0.

This is my compile option.

No problem.
[image]

MaxYoshida, 2023-01-21:
I have same compile options.

I tried to remove all hardware (sorry forgot to mention it), issue is the same.

This happens on more than one blue pill. I tried it on 2 of it.

If you believe it is still a hardware issue, I'll buy another 2 or 3 and give it a shot. Again, if this is the problem and it works, I'll be in your debt and pay you for it.

If you think we need a debug session, please tell me. Also, is there anything else you think about? Any other possible issues?

Do you think changing the whole MCU to something else will help? I mean, I can use a different type than a blue pill? It only needs to be cheap, available in like 1M piece in market and has CAN bus, ADC, PWM and fee IO pins.

nopnop2002, 2023-01-21:
Is this the same result?
```
  bool ret = CANInit(CAN_500KBPS, 0);  // CAN_RX mapped to PA11, CAN_TX mapped to PA12
  //bool ret = CANInit(CAN_500KBPS, 2);  // CAN_RX mapped to PB8, CAN_TX mapped to PB9
```

MaxYoshida, 2023-01-21:
I have tried with CANInit(CAN_500KBPS, 0);
It is what I was trying from beginning. Maybe I should try eith pins BP8 ans BP9?

I will check Linux option.

Which esp has can bus? Which model?

nopnop2002, 2023-01-21:
I noticed another possibility.
Isn't the clone used in your BluePill?

https://hackaday.com/2020/10/22/stm32-clones-the-good-the-bad-and-the-ugly/

Check the chip markings

nopnop2002, 2023-01-21:
>Which esp has can bus? Which model?

| Supported Targets | ESP32 | ESP32-C3 | ESP32-S2 | ESP32-S3 |

MaxYoshida, 2023-01-21:
OMG! Bro, you are amazing. I opened the link, and to my surprise I have the clone shown in 1st picture comparison. Is it possible that I spent a whole month trying with clones.. 

At this point I have no idea what to do, and how to be sure I am buying original piece and not a clone. I bought these from Amazon Deutschland.

About ESP, yes I heard about ESP-C3. I will give it a shot as well.

But as for now, I am more confident that I have a hardware issue as you pointed, because I am using a clone. Of course I will try to find original one and buy it.

May I ask if you have paypal? I really want to thank you so far

nopnop2002, 2023-01-21:
Good to know the cause.

MaxYoshida, 2023-01-21:
You seem not to be very open to paypal idea. So if there is anything I can do for you, just let me know.

Any ideas how to be sure I am buying an original blue pill and not a clone?

I am Japanese by origin. But currently in Germany for work.

nopnop2002, 2023-01-21:
>I am Japanese by origin.

I am Japanese. 
I live in Nagoya.
But you probably can't display Japanese because you don't have Japanese fonts installed on your PC.

>Any ideas how to be sure I am buying an original blue pill and not a clone?

Buy several from different sellers.

MaxYoshida, 2023-01-21:
You are right. I only have English and German keyboards here. Besides, I need to work more on my Japanese as the language dies when not used, and I left Osaka when I was 12.

I will take your advice and buy several from different sellers and see how it goes. I will also buy the esp32 as well.

I understand it is a hobby and not business, but at least there could be something I can do for you? Anything?

nopnop2002, 2023-01-21:
>I understand it is a hobby and not business, but at least there could be something I can do for you? Anything?

teach me german :)

MaxYoshida, 2023-01-21:
About German, I'd absolutely do that. We can have online sessions and I'll do it if this is what you want.
About cooking sausage, not something I can help with. I am vegetarian..

Hey, how about we talk using whatsapp? If you are open for it of course..

My number can be found inside this link: https://docs.google.com/document/d/1G0gJc9-uuI-XFvCi97k06_Ib1Zad2vJHtIrokvYT-P0/edit?usp=drivesdk

I'll delete it once you see it. If you are not interested, then I apologize and you can just ignore it

nopnop2002, 2023-01-21:
Unfortunately I can't have a conversation in English.

Same as most of Japanese.

MaxYoshida, 2023-01-21:
Demo, nihongo o tsukatte mo ī ndesu yo. Rōma ji de kaku to

nopnop2002, 2023-01-21:
>Demo, nihongo o tsukatte mo ī ndesu yo. Rōma ji de kaku to

Thnak you,

I will close this issue.

You can RE-OPEN at any time.

closed 2023-01-21 by nopnop2002

dartrax, 2023-02-23:
> I noticed another possibility. Isn't the clone used in your BluePill?
> 
> https://hackaday.com/2020/10/22/stm32-clones-the-good-the-bad-and-the-ugly/
> 
> Check the chip markings

Thanks for mentioning this. I had the same problem. But while my bluepill looks genuine, after further inspection I'm convinced that it's a fake without CAN functionality.

nopnop2002, 2023-02-23:
@dartrax 
Great to hear that.

nopnop2002, 2023-02-23:
@dartrax 

If you have time, it would be helpful for others if you could publish how to determine if it's a clone.

dartrax, 2023-02-23:
I used https://mecrisp-stellaris-folkdoc.sourceforge.io/bluepill-diagnostics-v1.6.html
It showed it was a clone. That alone might not mean that CAN doesn't work. But I rechecked the reviews of the shop where I bought it and I found at least one which confirmed that CAN functionality was missing. I also tried other libraries which failed at the same spot. Now I've got an STM32F072 so I'll try with that since I read that nearly all available Bluepills are fake now  ;-)

nopnop2002, 2023-02-28:
@dartrax 
Thank you for report.