nopnop2002/Arduino-STM32-CAN #53: Send Fail on STM32F407VE

bdamarjati, 2023-02-10:
i am using STM32F407VE discovery board with F_CPU 168MHZ, and SN65HVD230 CAN for the transceiver trying to send message from CAN1 (PD0, PD1) to CAN2(PB12,PB13) and get send fail output...
[image]

i am pretty sure the wiring is correct and there aren't any problem with CAN transceiver module, i also have changed the bit timing but still the same output, any solution or idea on whats the problem i have?

nopnop2002, 2023-02-10:
If the transmission fails, these are the possible causes.   
- There is no receiving app on CanBus.
- The speed does not match the receiver.
- There is no terminating resistor on the CanBus.
- There are three terminating resistors on the CanBus.
- The resistance value of the terminating resistor is incorrect.
- 3.3V transceiver and 5V transceiver are mixed.   
 +1V Ground Shift using Split Termination.   
 Check [here](http://www.ti.com/lit/an/slla337/slla337.pdf).   
- Stub length in CAN bus is too long.   
 Though the ISO11898 Standard specifies a maximum bus length of 40m and maximum stub length of 0.3m at 1Mbps.
 See [here](https://e2e.ti.com/support/interface-group/interface/f/interface-forum/378932/iso1050-can-bus-stub-length).   
- Noise effects.   
 It occurs when the wire cable is thin or wire cable is crossing.   
 It is preferable to use wires that are noise-proof.

bdamarjati, 2023-02-10:
i already double check the frequency, and it was 168mhz... i also change the clock to 72mhz the same as bluepil and still get send fail output...

i try the connection using bluepil it was running well, so i think there aren't any problem whatsoever in the conection and the receiver side

may i know what clock speed you use for the F407 one? maybe the problem was on the cpu clock

bdamarjati, 2023-02-10:
this is the STM32F407VE that i used
[image]

nopnop2002, 2023-02-10:
Are you using the Arduino IDE?
Or PlatformIO are you using?

bdamarjati, 2023-02-10:
im using arduino ide with stm32 core as you recommend

nopnop2002, 2023-02-10:
i want to see your build options
This is my build option.
[image]

bdamarjati, 2023-02-10:
mine are the same as yours but using serial as the upload method

nopnop2002, 2023-02-10:
it's definitely a wiring issue

bdamarjati, 2023-02-10:
the initializing is success but the transmitting is the issue, so i think the board can detect the transceiver module but fail when sending/receiving message

bdamarjati, 2023-02-10:
i try to receive packet from other can bus device with the same baudrate, and there aren't any message receive on STM32F407VE
[image]

nopnop2002, 2023-02-10:
One hint is that the sender did not detect the transmission failure.
A sender detects a send failure when it cannot send.
Try this with F103.

bdamarjati, 2023-02-13:
oh hi thank for the reply, i already checked using F103 send to F407, the output is Send Fail on F103 and blank on F407

the thing is, i check the wiring using F103 send to Arduino Nano is running well

any idea about this issue?

nopnop2002, 2023-02-13:
There have been many reports of F103 sending failures in the past.
Most of them can receive but fail to send.
If the transmission fails, these are the possible causes.   
- There is no receiving app on CanBus.
- The speed does not match the receiver.
- There is no terminating resistor on the CanBus.
- There are three terminating resistors on the CanBus.
- The resistance value of the terminating resistor is incorrect.
- 3.3V transceiver and 5V transceiver are mixed.   
 +1V Ground Shift using Split Termination.   
 Check [here](http://www.ti.com/lit/an/slla337/slla337.pdf).   
- Stub length in CAN bus is too long.   
 Though the ISO11898 Standard specifies a maximum bus length of 40m and maximum stub length of 0.3m at 1Mbps.
 See [here](https://e2e.ti.com/support/interface-group/interface/f/interface-forum/378932/iso1050-can-bus-stub-length).   
- Noise effects.   
 It occurs when the wire cable is thin or wire cable is crossing.   
 It is preferable to use wires that are noise-proof.   

I changed the single wire to twisted wire. It's fixed.   
[image]

bdamarjati, 2023-02-14:
okay thank for the reply, i managed to make it work somehow

turns out it was the can speed, possibly because of the signal frequency on F407 is higher or some kind of thing, so i used half can bus speed of the receiver

e.g: Arduino (receiver) using 500kbps on F407 (sender) using 250kbps and vice versa in order to make the transmission work, at least in my case

bdamarjati, 2023-02-14:
mine's already work so i think i'm going to close the issue now

thanks

closed 2023-02-14 by bdamarjati