nopnop2002/Arduino-STM32-CAN #54: how is Serial and CAN being used simultaneously in STM32F103?

i-make-robots, 2023-02-15:
STM32F103 manual says:

In low, medium-, high- and XL-density devices the USB and CAN share a dedicated 512-byte SRAM memory for data transmission and reception, and so they cannot be used concurrently (the shared SRAM is accessed through CAN and USB exclusively). The USB and CAN can be used in the same application but not at the same time.

so... how do I make sure they stay non-concurrent?

nopnop2002, 2023-02-15:
>The USB and CAN can be used in the same application but not at the same time.

Yes, that's right.

closed 2023-02-19 by nopnop2002

i-make-robots, 2023-02-27:
Thank you for the advice.  Is there a chip in the 48 pin package that would work?  I have been using STM32F103 which is 48 pin.
The STM32F411 looked good but it does not have CAN.  I am struggling to find the right match.