nopnop2002/Arduino-STM32-CAN #55: STM32F413ZH NUCLEO-F413ZH

pawkon, 2023-04-10:
Hi, any chance to adopt the code for stm32f413? To be honest Your code is the only one working for me with stm32f103 bluepill and I need to create CAN-BUS with F413 Nucleo. BTW great job!

nopnop2002, 2023-04-10:
The currently supported STM32F405/407 microcontrollers run at 168MHz.
The STM32F413/423 microcontrollers run at 100MHz.
This difference requires changing this value for STM32F413/423 microcontrollers.

https://github.com/nopnop2002/Arduino-STM32-CAN/blob/master/stm32f407/stm32f407.ino#L29

Since I don't have an STM32F413/423 microcontroller,
I can't show you how to change it.

pawkon, 2023-04-11:
There is a site : http://www.bittiming.can-wiki.info/ on which we could calculate the timing but I have no idea what the digits mean in Your code. Can You describe which is doing what?

For example 50MHz bittiming should be like this:

`{50,10,0x000701f3}, {50,20,0x000700f9}, {50,50,0x00070063}, {50,100,0x00070031}, {50,125,0x001c0018}, {50,250,0x00070013}, {50,500,0x00070009}, {50,1000,0x00070004},`

systemclock of the f413 is 100MHz but not the CAN? As on the site: "Use the value of the clock rate at the first stage of the BaudRatePrescaler BTR, not the clock of the controller or crystal (typically for a 16 MHz clocked NXP SJA1000 use '8')" Is it the same for ST?

Is this the only one parameter to change in Your code?

nopnop2002, 2023-04-11:
These two codes set the BTR register.
STM32 uses three values ​​TS2, TS1 and BRP to configure the BTR register.

Examine these three values ​​and set them in the can_configs variable.

https://github.com/nopnop2002/Arduino-STM32-CAN/blob/master/stm32f407/stm32f407.ino#L220

https://github.com/nopnop2002/Arduino-STM32-CAN/blob/master/stm32f407/stm32f407.ino#L224

>Is this the only one parameter to change in Your code?

You should read the reference manual carefully.

Look up power control in the RCC chapter.

Find out how to configure the CAN port in the GPIO chapter.

Find out how to use filters in the bxCAN chapter.

And reflect the result of the examination in the code.

https://www.st.com/resource/en/reference_manual/rm0430-stm32f413423-advanced-armbased-32bit-mcus-stmicroelectronics.pdf

Here is the STM32413ZH datasheet.

You should read this to find out which port's alternate function mapping CAN uses.

https://www.alldatasheet.com/datasheet-pdf/pdf/885152/STMICROELECTRONICS/STM32F413ZH.html

It is hard work.

pawkon, 2023-04-11:
@nopnop2002 is there a way I could write You a privet message?

nopnop2002, 2023-04-11:
By arranging a time with you, I can publish my private address here.

I will delete the message as soon as you read it.

pawkon, 2023-04-12:
Perfect. I'm waiting for Your email then

pawkon, 2023-04-12:
yes :)

nopnop2002, 2023-04-12:
Reveal for 1 minute

hau26280@tree.odn.ne.jp

pawkon, 2023-04-12:
got it

nopnop2002, 2023-04-12:
Please send me an email with just the title to confirm that the address is correct.

pawkon, 2023-04-12:
I've already sent You with the title of this issue and info.

nopnop2002, 2023-04-12:
I received your email.

nopnop2002, 2023-09-09:
I added.

closed 2023-09-09 by nopnop2002