nopnop2002/Arduino-STM32-CAN #58: Mistake in Line 277

KGBRUS1, 2023-05-10:
Dear Nopnp2002,

i think there is a mistake in Line 277 " GPIOA->ODR |= 0x1UL << 12;       // PA12 Upll-up"

The PA12 is the TX pin, i think that the RX pin (PA11) must be usesed in this line

simulat to  GPIOB->ODR |= 0x1UL << 8;        // PB8 Upll-up (this is RX Pin

best regards

Kay

by the way, thanks for your Sketch!

KGBRUS1, 2023-05-10:
Sorry:

[Arduino-STM32-CAN](https://github.com/nopnop2002/Arduino-STM32-CAN/tree/master)/[stm32f103](https://github.com/nopnop2002/Arduino-STM32-CAN/tree/master/stm32f103)
/stm32f103.ino

nopnop2002, 2023-05-10:
It looks like a bug.
Test then fix.
Thanks for the report.

KGBRUS1, 2023-05-10:
Thanks for your excelent work! i never get CAN and Blue Pill running without your sketch!

nopnop2002, 2023-05-13:
Fixed.

thank you.

closed 2023-05-13 by nopnop2002