nopnop2002/Arduino-STM32-CAN #60: feature request: support multiple transmit mailboxes

i-make-robots, 2023-08-25:
Currently sTxMailBox array appears to be hard coded to use box 0, tho (on the STM32F405) there are at least 3.  I'd love to see some code that identifies an available mailbox and uses that, or warns if there are none available.

nopnop2002, 2023-08-25:
# Advanced use of BxCan   
Skip if you don't need advanced usage.   
Advanced usage has different purposes depending on the requirements of the application.   
So I can't give you a concrete code example.   

### About transmission
STM32 has three transmit(TX) mailboxes.   
Three TX mailboxes are provided to the software for setting up messages.   
This sample code uses only the first TX mailbox and blocks the application until the send is successfully completed.   
There are two ways to know if the send was successful without blocking.   
Read the reference manual carefully and modify the source code as needed.   

- Uses a transmission completion interrupt   
 You need to set the CAN interrupt enable register (CAN_IER) appropriately.   

- Check TX mailbox space before sending   
 An application can store transmission requests in three TX mailboxes simultaneously.   
 When there are requests for multiple TX mailboxes, the transmission scheduler decides which mailbox has to be transmitted first.   
 You can specify which mailbox has priority using the CAN Master Control Register (CAN_MCR).   
 You can tell if a mailbox is free using the CAN Transmit Status Register (CAN_TSR).   
 When all three TX mailboxes are pending, no new messages can be sent until the transmit is canceled or completed.   
 When all three TX mailboxes are pending, you can choose to cancel the pending transmission, wait until the transmission is complete, or give up on the new transmission.   
 This choice depends on your application requirements.   
 When all three TX mailboxes are pending for a long time, it could be a transmission error.   
 CAN Error Status Register (CAN_ESR) should be checked.   

### About reception
STM32 has two receive(RX) mailboxes.   
Each RX Mailbox allows access to a 3-level depth FIFO, the access being offered only to the oldest received message in the FIFO.   
This sample code uses only the first RX mailbox.   
The RX mailbox is closely related to the receive filter settings.   
By properly setting the receive filter, you can sort the received messages into two RX Mailboxes.   
For example, high priority messages can be stored in the first RX MailBox and low priority messages can be stored in the second RX MailBox.   
Read the reference manual carefully and modify the source code as needed.

i-make-robots, 2023-08-25:
Thank you!  That seems to match what I read and gives me confidence to proceed.

nopnop2002, 2023-09-18:
First of all, to write a conclusion, you cannot use bxCan interrupts with Arduino core support for STM32.

The interrupt should work with the following steps, but it will HALT.

Add this in CANInit()

```
  CAN1->IER  |=   0x1UL;                // Set to Transmit mailbox empty interrupt enable
```

Add this in Setup()

```
  NVIC_SetPriority(CAN1_TX_IRQn, 1);
  NVIC_EnableIRQ(CAN1_TX_IRQn);
```

Transmit interrupts are now enabled.

The interrupt handler is defined with the name CAN1_TX_IRQHandler().
This is a week function, so you can replace it.

```
extern "C" void CAN1_TX_IRQHandler (void)
{
  Serial.println("USB_HP_CAN1_TX_IRQHandler");  
}
```

This interrupt function is called when transmission is completed, but when this function ends, it becomes HALT.

It has been reported to STM, but no countermeasures have been taken yet.

i-make-robots, 2023-09-19:
Interesting!  Here's what I did https://github.com/MarginallyClever/Daisy-Driver-2.0-firmware/commit/465ec3dbeb831ee2f49fd6a6648da8fd7377a56a

in Arduino 2.2.1 with https://github.com/stm32duino/BoardManagerFiles/raw/master/package_stmicroelectronics_index.json

```c
void CANInit() {
  // ...

  // attach message pending interrupt method
  NVIC_SetPriority(CAN1_RX_IRQn, 1);
  NVIC_EnableIRQ(CAN1_RX_IRQn);

  // Enable FIFO Message Pending Interrupt 
  CAN1->IER |= CAN_IER_FMPIE0 | CAN_IER_FMPIE1;
  DEBUGLN("CAN1 interrupt enabled.");

  return true;
}

int CANstate2,CANstate1;
void CAN1_RX0_IRQHandler(void) {
  light.setColor(0,CANstate1,CANstate2);
  CANstate2 = (CANstate2==0? 255 : 0);
  CAN1->RF0R |= CAN_RF0R_RFOM0;  // release FIFO
  CAN1->IER |= CAN_IER_FMPIE0;  // enable interrupt
}

void CAN1_RX1_IRQHandler(void) {
  light.setColor(0,CANstate2,CANstate2);
  CANstate1 = (CANstate1==0? 255 : 0);
  CAN1->RF1R |= CAN_RF1R_RFOM1;  // release FIFO
  CAN1->IER |= CAN_IER_FMPIE1;  // enable interrupt
}
```
I confirm this problem case.  No response from STM tells me nobody write the firmware this way because 
a) stm32cubeide is what most people use or 
b) polling should be enough or 
c) almost nobody writes CAN stuff for this board.

nopnop2002, 2023-09-19:
Does light.setColor() run correctly?

nopnop2002, 2023-09-20:
You need to change:

//void CAN1_RX0_IRQHandler(void) {
extern "C" void CAN1_RX0_IRQHandler(void) {

But don't work.

i-make-robots, 2023-09-21:
It hated external “c”Sent from my iPhone

i-make-robots, 2023-10-02:
Any idea why HALT occurs?

nopnop2002, 2023-10-02:
Probably a bug in the core library.