nopnop2002/Arduino-STM32-CAN #61: BARTH STG-850 Support

ben-mcdonnell-hacksmith, 2023-09-19:
I am trying to use your library with a [BARTH STG-850](https://barth-elektronik.com/en/lococube-mini-PLCs/lococube-mini-PLC-STG-850.html) using their [core](https://github.com/jasysdotde/STG-8xx). The library does build, but I am having issues sending messages. I have run through the troubleshooting steps in your README. I have termination resistors on both ends of the wire (150 ohm) and the device doesn't have any internal termination. 

The PLC is STM32F09x based. I am not sure what the internal CAN transceiver is, but it does use PB8/9 for the pins, and I configured the software as such. 

When starting I do get the CAN Initialization OK message.

GPIO_AFR(b)=0x0
GPIO_AFR(a)=0x4
GPIO_MODER(b)=0x0
GPIO_MODER(a)=0x20000
GPIO_OSPEEDR(b)=0x0
GPIO_OSPEEDR(a)=0x30000
GPIO_OTYPER(b)=0x0
GPIO_OTYPER(a)=0x0
GPIO_PUPDR(b)=0x0
GPIO_PUPDR(a)=0x0
GPIO_AFR(b)=0x4
GPIO_AFR(a)=0x44
GPIO_MODER(b)=0x20000
GPIO_MODER(a)=0xA0000
GPIO_OSPEEDR(b)=0x30000
GPIO_OSPEEDR(a)=0xF0000
GPIO_OTYPER(b)=0x0
GPIO_OTYPER(a)=0x0
GPIO_PUPDR(b)=0x0
GPIO_PUPDR(a)=0x0
CAN->BTR=0x1E0003
CAN1 initialize ok
Send Fail
352321621
3080
419430408
Send Fail
1962934357
3080
419430408
Send Fail
788529237
3080
419430408

Any suggestions?

Thanks!

nopnop2002, 2023-09-19:
>I am not sure what the internal CAN transceiver is

You need to clarify the transceiver you are using.

ben-mcdonnell-hacksmith, 2023-09-20:
They aren't super forthcoming with details, but from some Keil example code, it appears to be a sn65hvd234 transceiver.

nopnop2002, 2023-09-20:
Many cases of Send Fail have been reported in the past.

Both are caused by an inappropriate environment.

ben-mcdonnell-hacksmith, 2023-09-29:
The board isn't running at 48mhz, according to the datasheet it is running at 16mhz, would I need to adjust the CAN timing to fix that?

nopnop2002, 2023-09-29:
>would I need to adjust the CAN timing to fix that?

Yes.