nopnop2002/Arduino-STM32-CAN #62: Unable to communicate STM32f103 vs ESP32 via CAN BUS

thangduc131199, 2023-12-28:
I have 1 circuit STM32F103 Blue Pill and ESP32, 2 SN65HVD23x CAN Transceiver, I changed the resistor to 120 ohm -> 150 ohm but still cannot communicate via CAN bus. For STM32 I use the HAL library, and for ESP32 I use the CAN library https://github.com/sandeepmistry/arduino-CAN. Hope you can give me some suggestions to solve the problem. Thanks
[image]

thangduc131199, 2023-12-28:
I had an unexpected discovery. When I set the Bitrate of ESP32 to 1000kbs and STM32 to 500kbs, the two circuits can communicate with each other via CAN bus. However, checking that there are still incorrect messages and the CAN controler sends them again (I sent the message at 10hz and only received it at 3hz).

nopnop2002, 2023-12-28:
If you specify Rev2 or higher for the ESP32 Revision, the CAN BitRate will be half of the specified value.
[image]

See below for details.

https://www.esp32.com/viewtopic.php?t=15581

closed 2024-10-01 by nopnop2002