nopnop2002/Arduino-STM32-CAN #63: Sending data issue

banbeiduyao, 2024-03-20:
I am currently using one F405, one F103, and USB to CAN for testing. There is no problem transferring data between USB to CAN and F103. However, after connecting F405 to send data for a period of time, F103 cannot send data. The reason cannot be determined!
Screenshots of normal and abnormal situations
[image]
[image]

nopnop2002, 2024-03-20:
What exactly is USB to CAN?

banbeiduyao, 2024-03-20:
CAN data can be monitored through a computer

nopnop2002, 2024-03-20:
It works correctly when there are two nodes, USB to CAN and F103.

However, it does not work correctly when there are three nodes: USB to CAN, F103, and F405.

Is this understanding correct?

banbeiduyao, 2024-03-20:
Yes, as long as F405 is connected, F103 will fail to send data after a period of time, and the time may not be certain

nopnop2002, 2024-03-20:
It works correctly when there are two nodes, USB to CAN and F103. ---> CAN-BUS length is short.

It works correctly when there are two nodes, USB to CAN and F405. ---> CAN-BUS length is short.

It works correctly when there are two nodes, F103 and F405. ---> CAN-BUS length is short.

However, it does not work correctly when there are three nodes: USB to CAN, F103, and F405. ---> The length of CAN-BUS is a little longer.

---

The CAN bus stub length may be too long.
Though the ISO11898 Standard specifies a maximum bus length of 40m and maximum stub length of 0.3m at 1Mbps. See [here](https://e2e.ti.com/support/interface-group/interface/f/interface-forum/378932/iso1050-can-bus-stub-length).

banbeiduyao, 2024-03-20:
Thank you very much, but the wiring distance here is very short. F405>>5cm>>F103>>20cm>>USB-CAN

nopnop2002, 2024-03-20:
If you have a mix of 3.3V and 5V transceivers, a +1V ground shift using split termination is very beneficial.
Check here.
https://www.ti.com/lit/an/slla337/slla337.pdf