nopnop2002/Arduino-STM32-CAN #64: dlc bigger than 8

gravefunc, 2024-04-30:
“Hi, I’m using an STM32F103 microcontroller, and I’m encountering an issue with the Data Length Code (DLC) exceeding 8.”
Extended ID: 0x09F20009 DLC: 8 Data: 0x1 0x70 0x3 0xFF 0xFF 0x7F 0xFF 0xFF<< this one priority is 5 and correct data is =ff 01 0e 6f 9f ff ff ff

Extended ID: 0x09F20109 DLC: 8 Data: 0x60 0x1A 0x0 0xD0 0x7 0xEB 0x10 0x87<<dlc is 26 and data corrupted
Extended ID: 0x09F20109 DLC: 8 Data: 0x61 0x82 0xFF 0x7F 0xFF 0x7F 0xFF 0xFF
Extended ID: 0x15FD0C09 DLC: 8 Data: 0xFF 0x1 0xE 0x56 0x3A 0x6 0xFF 0xFF

Extended ID: 0x09F20009 DLC: 8 Data: 0x1 0x78 0x3 0xFF 0xFF 0x7F 0xFF 0xFF<<dlc is 8 and data is correct

nopnop2002, 2024-04-30:
It's not a clone chip, right?

https://hackaday.com/2020/10/22/stm32-clones-the-good-the-bad-and-the-ugly/

gravefunc, 2024-04-30:
`Extended` ID:9F20009 DLC: 8 buf: 0x1 0x7F 0x3 0xFF 0xFF 0x7F 0xFF 0xFF
Extended ID:9F20109 DLC: 8 buf: 0xE0 0x1A 0x0 0xD0 0x7 0xEB 0x10 0x87
Extended ID:9F20109 DLC: 8 buf: 0xE1 0x82 0xFF 0x7F 0xFF 0x7F 0xFF 0xFF
Extended ID:9F20109 DLC: 8 buf: 0xE2 0xFF 0xFF 0x0 0x0 0xFF 0xFF 0xFF
Extended ID:9F20109 DLC: 8 buf: 0xE3 0x0 0x0 0x0 0x0 0x7F 0x7F 0xFF
Extended ID:15FD0809 DLC: 8 buf: 0xFF 0x1 0xE 0x6F 0x9F 0xFF 0xFF 0xFF
Extended ID:15FD0C09 DLC: 8 buf: 0xFF 0x1 0xE 0x56 0x3A 0x6 0xFF 0xFF
Extended ID:9F20009 DLC: 8 buf: 0x1 0x80 0x3 0xFF 0xFF 0x7F 0xFF 0xFF`

the correct one is somthing like this which is idf201 still have some extra data im working on it

nopnop2002, 2024-04-30:
I want to see the sender log.

gravefunc, 2024-04-30:
IM USING https://github.com/ttlappalainen/NMEA2000 FOR ESP32 AND compare that output with yours in stm

nopnop2002, 2024-04-30:
Your STM32F103 is not a clone, is it?

https://hackaday.com/2020/10/22/stm32-clones-the-good-the-bad-and-the-ugly/

gravefunc, 2024-05-01:
5157828 : Pri:2 PGN:127488 Source:9 Dest:255 Len:8 Data:1,82,3,FF,FF,7F,FF,FF
5157839 : Pri:2 PGN:127489 Source:9 Dest:255 Len:26 Data:0,D0,7,EB,10,87,82,FF,7F,FF,7F,FF,FF,FF,FF,0,0,FF,FF,FF,0,0,0,0,7F,7F
5157882 : Pri:5 PGN:130312 Source:9 Dest:255 Len:8 Data:FF,1,E,6F,9F,FF,FF,FF
5157903 : Pri:5 PGN:130316 Source:9 Dest:255 Len:8 Data:FF,1,E,56,3A,6,FF,FF

closed 2024-05-10 by nopnop2002