nopnop2002/Arduino-STM32-CAN #65: Support for working with stm32f411 and changing contacts PA11 and PA12

brightproject, 2024-09-27:
Hello @nopnop2002 .
I see that your wonderful library supports `stm32f103` and `stm32f413` microcontrollers.
I have `stm32f411ceu6` microcontroller and `mcp2551` CAN xcvr

[image]

Can I run the code on it?
And can I change the contacts, from `PA11 PA12` to `PB8 PB9`?
I have never worked with `CAN` bus before, so the questions are very stupid.
Maybe it makes sense to open discussions in your repository, so as not to post such questions in the issue?
In the code assembly I use:

> https://github.com/stm32duino/Arduino_Core_STM32

and `vscode+prlatformio`.

nopnop2002, 2024-09-27:
STM32F411 does not have CAN functionality.

https://www.st.com/ja/microcontrollers-microprocessors/stm32f411/products.html

brightproject, 2024-09-28:
> When using the CAN function with STM32, the MCU must be selected carefully.

Thanks for the clarification, quite annoying.
But I will use the module temporarily:

[image]

closed 2024-09-28 by brightproject

nopnop2002, 2024-09-28:
The CAN transceiver (TJA1050) used in this module is a 5V power supply, so
For 3.3V machines such as STM32, SPI level shifting is required.

nopnop2002, 2024-09-28:
Not all GPIOs of STM32 are 5V tolerant.

Please note that this module is 5V powered, so you will need to use a 5V tolerant GPIO.

brightproject, 2024-09-29:
> I replaced the transceiver chip

You are great that you suggested this🙂
And did you change the quartz from `8 kHz` to something with a `higher` frequency?
I have chips `SN65HVD230`, I also thought about soldering them onto the module `MCP2515`.

nopnop2002, 2024-09-29:
>did you change the quartz from 8 kHz to something with a higher frequency?

No.

brightproject, 2024-10-20:
> No.

Please tell me to use the board with the controller and transceiver `MCP2515` + `SN65HVD230`

[image]

And `stm32411ceu6` via the `SPI` bus

What code should I use from your repo?

All examples, including this one

> https://github.com/nopnop2002/Arduino-STM32-CAN/tree/master/stm32f103

do not use the `SPI` bus for connection.

nopnop2002, 2024-10-20:
STM32 includes CAN controller functionality.
Therefore, MCP2515 is not used.

brightproject, 2024-11-10:
> However, I have never used this library with STM32F411

Yes, I have tried [this](https://github.com/coryjfowler/MCP_CAN_lib/blob/master/mcp_can.cpp) library, as well as [another](https://github.com/Longan-Labs/Arduino_CAN_BUS_MCP2515/blob/master/mcp_canbus.cpp) one that is more versatile.
Both libraries in codes work +/- the same.
I'm trying this simple code example
```
#include <Arduino.h>
#include <mcp_can.h>

// CAN
const int CAN_CS_PIN = PA4;

// int8_t Airspeed = 0;
// int32_t Airspeed = 0;
  unsigned int Airspeed = 0;
// unsigned long Airspeed = 0;
bool countup[1];

unsigned char canMsg[8] = {0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07};

const unsigned int CAN_ID = 183; // CAN Msg ID in DEC
const unsigned int Period = 100; // How often message sent in milliseconds
unsigned long Timestamp = 0; // when was the last message sent

MCP_CAN CAN(CAN_CS_PIN);  // Set CS pin

void setup()
{
 Serial.begin(115200);

    // while (CAN_OK != CAN.begin(MCP_ANY, CAN_500KBPS, MCP_8MHZ)) { // init can bus : baudrate = 500k
    while(CAN.begin(MCP_ANY, CAN_125KBPS, MCP_8MHZ) != CAN_OK) {
    // while(CAN_OK != CAN.begin(CAN_125KBPS, MCP_8MHZ)) {
      Serial.println("Unable to begin CAN BUS");
      delay(1000);
    }
    CAN.setMode(MCP_NORMAL);
    Serial.println("CAN BUS Shield init ok!");
  
} // SETUP

void loop()
{

if (millis() > Timestamp + Period) {

  if (Airspeed <= 100) {
    countup[0] = true;
  } 
  
  else if (Airspeed >= 1000) {
    countup[0] = false;
  }

  if (countup[0]) Airspeed += 5;
  else Airspeed -= 5;

  // canMsg[0] = Airspeed;
  // canMsg[1] = Airspeed >> 8;
  // canMsg[2] = Airspeed >> 16;
  // canMsg[3] = Airspeed >> 24;

  canMsg[0] = 1;
  canMsg[1] = 2;
  canMsg[2] = 3;
  canMsg[3] = 4;

 Serial.print(canMsg[0], HEX);
  Serial.print(".");
  Serial.print(canMsg[1], HEX);
  Serial.print(".");
  Serial.print(canMsg[2], HEX);
  Serial.print(".");
  Serial.println(canMsg[3], HEX);

  CAN.sendMsgBuf(CAN_ID, 0, 8, canMsg); 
  
  Timestamp = millis();
}

} // LOOP
```

The serial port data goes as expected.

> 4E.2.0.0
> 49.2.0.0
> 44.2.0.0
> 3F.2.0.0
> 3A.2.0.0
> 35.2.0.0
> 30.2.0.0
> 2B.2.0.0
> 26.2.0.0
> 21.2.0.0

Do you know what might be the reason?

There's complete nonsense with the can, the first frame sends 15 and that's it, then it freezes in all frames.

One
[image]
Two
[image]

I read somewhere that the `MCP2515` modules have an `8 MHz` quartz crystal and it doesn't work with some library normally, maybe I'm in exactly this situation.
I'll try to solder a `16 MHz` quartz crystal or use another module.

nopnop2002, 2024-11-10:
Arduino_Core_STM32 supports SPI library, but only works in Transaction SPI Mode.

- Legacy SPI Mode   
SPI speed depends on the CPU frequency.   
STM32 does not work because it has various CPU frequencies.
```
SPI.begin()
SPI.setClockDivider(divider)
SPI.setDataMode(mode)
SPI.setBitOrder(order)
SPI.transfer(data);
```

- Transaction SPI Mode   
SPI speed NOT depends on the CPU frequency.   
Therefore, it works correctly with STM32.
```
SPI.begin()
SPI.beginTransaction(SPISettings(speed, order, mode))
SPI.transfer(data);
SPI.endTransaction()
```

nopnop2002, 2024-11-12:
Transmission was possible with STM32F411 + mcp_can.
However, since there is no INT pin, reception is not possible.
```
// CAN Send Example
//

#include <mcp_can.h>
#include <SPI.h>

//MCP_CAN CAN0(10);     // Set CS to pin 10
MCP_CAN CAN0(PA4);     // Set CS to pin 10

//#define SPEED CAN_125KBPS
//#define SPEED CAN_250KBPS
#define SPEED CAN_500KBPS
//#define SPEED CAN_1000KBPS

void setup()
{
  Serial.begin(115200);
  Serial.print("CAN speed=");
  Serial.println(SPEED);

  // Initialize MCP2515 running at 16MHz with a baudrate of 125kb/s and the masks and filters disabled.
  if(CAN0.begin(MCP_STDEXT, SPEED, MCP_8MHZ) == CAN_OK) {
  //if(CAN0.begin(MCP_ANY, SPEED, MCP_8MHZ) == CAN_OK) {
    Serial.println("MCP2515 Initialized Successfully!");
  } else {
    Serial.println("Error Initializing MCP2515...");
    while(1);
  }
  CAN0.setMode(MCP_NORMAL);   // Change to normal mode to allow messages to be transmitted
}

void loop()
{
  byte frame[8] = {0x10, 0x11, 0x12, 0x13, 0x14, 0x15, 0x16, 0x17};
  static uint8_t sendLen = 0;
  // send data:  ID = 0x100, Standard CAN Frame, Data length = 8 bytes, 'data' = array of data bytes to send

  static unsigned long extID = 0x10000100;

  byte sndStat = CAN0.sendMsgBuf(extID, 1, sendLen, frame);
  if(sndStat == CAN_OK){
    Serial.print("Message Sent Successfully! ");
    Serial.println(extID);
  } else {
    Serial.println("Error Sending Message...");
  }

  extID++;
  if (extID > 0x1000010F) extID = 0x10000100;
  sendLen++;
  if (sendLen > 8) sendLen = 0;
  
  delay(1000);   // send data per 100ms
}
```

brightproject, 2024-11-13:
This is great news, thank you @nopnop2002 
Please tell me how you debug the `CAN` bus, what equipment do you use to check sending and receiving messages, in addition to outputting to the serial port?
Could you please check this code?

```
#include <Arduino.h>
#include <mcp_can.h>

// CAN
const int CAN_CS_PIN = PA4;

// int8_t Airspeed = 0;
// int32_t Airspeed = 0;
  unsigned int Airspeed = 0;
// unsigned long Airspeed = 0;
bool countup[1];

unsigned char canMsg[8] = {0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07};

const unsigned int CAN_ID = 183; // CAN Msg ID in DEC
const unsigned int Period = 100; // How often message sent in milliseconds
unsigned long Timestamp = 0; // when was the last message sent

MCP_CAN CAN(CAN_CS_PIN);  // Set CS pin

void setup()
{
 Serial.begin(115200);

    // while (CAN_OK != CAN.begin(MCP_ANY, CAN_500KBPS, MCP_8MHZ)) { // init can bus : baudrate = 500k
    while(CAN.begin(MCP_ANY, CAN_125KBPS, MCP_8MHZ) != CAN_OK) {
    // while(CAN_OK != CAN.begin(CAN_125KBPS, MCP_8MHZ)) {
      Serial.println("Unable to begin CAN BUS");
      delay(1000);
    }
    CAN.setMode(MCP_NORMAL);
    Serial.println("CAN BUS Shield init ok!");
  
} // SETUP

void loop()
{

if (millis() > Timestamp + Period) {

  if (Airspeed <= 100) {
    countup[0] = true;
  } 
  
  else if (Airspeed >= 1000) {
    countup[0] = false;
  }

  if (countup[0]) Airspeed += 5;
  else Airspeed -= 5;

  // canMsg[0] = Airspeed;
  // canMsg[1] = Airspeed >> 8;
  // canMsg[2] = Airspeed >> 16;
  // canMsg[3] = Airspeed >> 24;

  canMsg[0] = 1;
  canMsg[1] = 2;
  canMsg[2] = 3;
  canMsg[3] = 4;

 Serial.print(canMsg[0], HEX);
  Serial.print(".");
  Serial.print(canMsg[1], HEX);
  Serial.print(".");
  Serial.print(canMsg[2], HEX);
  Serial.print(".");
  Serial.println(canMsg[3], HEX);

  CAN.sendMsgBuf(CAN_ID, 0, 8, canMsg); 
  
  Timestamp = millis();
}

} // LOOP
```

I check sending to `CAN` bus using `logic analyzer`, and the values ​​in it do not increase, they are stopped at `15`.

[image]

[image]

And this is either a signal analyzer problem or a `CAN` transmitter problem.
I tried this solution

> https://github.com/latonita/arduino-canbus-monitor

But I could not get it to work.

nopnop2002, 2024-11-13:
- There is no receiving app on CanBus.
- The speed does not match the receiver.
- There is no terminating resistor on the CanBus.
- There are three terminating resistors on the CanBus.
- The resistance value of the terminating resistor is incorrect.
- 3.3V transceiver and 5V transceiver are mixed.   
 Use +1V Ground Shift using Split Termination.   
 Check out Figure 7 [here](http://www.ti.com/lit/an/slla337/slla337.pdf).   
- Stub length in CAN bus is too long.   
 ISO11898 Standard specifies a maximum bus length of 40m and maximum stub length of 0.3m at 1Mbps.
 See [here](https://e2e.ti.com/support/interface-group/interface/f/interface-forum/378932/iso1050-can-bus-stub-length).   
- Noise effects.   
 It occurs when the wire cable is thin or wire cable is crossing.   
 It is preferable to use wires that are noise-proof.

brightproject, 2024-11-14:
> * Check out Figure 7 [here](http://www.ti.com/lit/an/slla337/slla337.pdf).

Thank you, I will study the material.
I am surprised that the `3.3. volt` transceiver has a slight imbalance of the two signals😞

[image]

It turns out that this event can also complicate the transmission of data over the `CAN` bus using the `SN65HVD230` transceiver?
Your code works for me, and my code works too.
I figured it out yesterday, and it turned out that my `Saleao` signal analyzer does not display the change of the date field on the `CAN` bus.

nopnop2002, 2024-11-14:
The GPIOs (PA4, 5, 6, 7) used by SRM32F411's SPI are all 5 V tolerant I/Os, so they operate with 5 V power supply.